sunspec/pysunspec #7: Can't speak to SMA Sunny Island device

bbinet, 2016-03-22:
I would like to use the pysunspec module to retrieve data from a SMA Sunny Island, but it does not seem to work. I get the following message when using the `suns.py` script:
```
$ suns.py -i 192.168.1.77
Error: Device responded - not SunSpec register map
```
So it says that the Sunny Island is not a SunSpec device, whereas the Sunny Island is supposed to speak the SunSpec modbus protocol, according to the SMA website: http://www.sma.de/en/products/battery-inverters/sunny-island-60h-80h.html#Downloads-206198

Any idea why it does not work?

bobfox, 2016-03-22:
The SMA Modbus register map at slave id 1 is not SunSpec compliant. SMA 
typically provides the SunSpec Modbus map at slave id 126 if it is 
available. If slave id 126 does not work, consult SMA to determine if 
the device you are communicating with supports the SunSpec interface.

bbinet, 2016-03-22:
Thanks, it works perfectly with slave id 126:
```
$ suns.py -i 192.168.1.77 -a 126

Timestamp: 2016-03-22T15:00:18Z

model: Common (1)

      Manufacturer (Mn):                                  SMA           
      Model (Md):                              Solar Inverter           
      Options (Opt):                                     9224           
      Version (Vr):                                  16974084           
      Serial Number (SN):                          1261038476           

model: Ethernet Link Layer (11)

      Ethernet Link Speed (Spd):                          100 Mbps      
      Interface Status Flags (CfgSt):                  0x0003           
      Link State (St):                                      1           
      MAC (MAC):                            00:40:AD:9A:FA:BF           

model: IPv4 (12)

      Config Status (CfgSt):                                1           
      Change Status (ChgSt):                           0x0000           
      Config Capability (Cap):                         0x0005           
      IPv4 Config (Cfg):                                    1           
      Control (Ctl):                                        0           
      IP (Addr):                                192.168.0.172           
      Netmask (Msk):                            255.255.255.0           
      Gateway (Gw):                                   0.0.0.0           
      DNS1 (DNS1):                                    0.0.0.0           

model: Inverter (Three Phase) (103)

      Phase Voltage AN (PhVphA):                        230.0 V         
      Phase Voltage BN (PhVphB):                          0.0 V         
      Phase Voltage CN (PhVphC):                          0.0 V         
      Watts (W):                                       -100.0 W         
      Hz (Hz):                                           48.9 Hz        
      VAr (VAr):                                       -200.0 var       
      Event1 (Evt1):                               0x00000000           

model: Nameplate (120)

      DERTyp (DERTyp):                                     82           
      VArRtgQ2 (VArRtgQ2):                                0.0 var       
      VArRtgQ3 (VArRtgQ3):                                0.0 var       
      PFRtgQ1 (PFRtgQ1):                                 -0.8 cos()     
      PFRtgQ2 (PFRtgQ2):                                  0.0 cos()     
      PFRtgQ3 (PFRtgQ3):                                  0.0 cos()     
      PFRtgQ4 (PFRtgQ4):                                  0.8 cos()     
      WHRtg (WHRtg):                                  11500.0 Wh        
      AhrRtg (AhrRtg):                                    240 AH        
      MaxChaRte (MaxChaRte):                           6000.0 W         
      MaxDisChaRte (MaxDisChaRte):                     6000.0 W         

model: Basic Settings (121)

      ECPNomHz (ECPNomHz):                                 50 Hz        

model: Measurements_Status (122)

      PVConn (PVConn):                                 0x0001           
      ECPConn (ECPConn):                               0x0000           

model: Immediate Controls (123)

      WMaxLim_Ena (WMaxLim_Ena):                            1           
      OutPFSet_Ena (OutPFSet_Ena):                          0           
[... 32 lines cut ...]
model: Watt-PF (131)

      ActCrv (ActCrv):                                      0           
      ModEna (ModEna):                                 0x0000           
      NPt (NPt):                                            2           
   01:ActPt (ActPt):                                        2           
   01:W1 (W1):                                             50 % WMax    
   01:W2 (W2):                                            100 % WMax    
   01:ReadOnly (ReadOnly):                                  0           

model: Volt-Watt (132)

      ActCrv (ActCrv):                                      1           
      ModEna (ModEna):                                 0x0001           
      NCrv (NCrv):                                          1           
      NPt (NPt):                                           12           
   01:DeptRef (DeptRef):                                    1           
   01:ReadOnly (ReadOnly):                                  0           

model: Multiple MPPT Inverter Extension Model (160)

      Number of Modules (N):                                2           
   01:Input ID (ID):                                        1           
   02:Input ID (ID):                                        2
```

closed 2016-03-22 by bbinet