sunspec/pysunspec #8: Tests are failing

bbinet, 2016-03-22:
When I try to run the tests on Linux (Debian) laptop, I get the following error:
```
$ pysunspec_test.py
pySunSpec version: 1.0.7
Test device path: /home/bruno/.virtualenvs/tmp-d3111b3cabc0fa7b/lib/python2.7/site-packages/sunspec/core/test/devices

Traceback (most recent call last):
  File "/home/bruno/.virtualenvs/tmp-d3111b3cabc0fa7b/bin/pysunspec_test.py", line 13, in <module>
    test.test_all()
  File "/home/bruno/.virtualenvs/tmp-d3111b3cabc0fa7b/local/lib/python2.7/site-packages/sunspec/core/test/test_all.py", line 42, in test_all
    (count_run, count_passed, count_failed) = module.test_all(pathlist, stop_on_failure)
  File "/home/bruno/.virtualenvs/tmp-d3111b3cabc0fa7b/lib/python2.7/site-packages/sunspec/core/test/test_device.py", line 561, in test_all
    if test(pathlist) is True:
  File "/home/bruno/.virtualenvs/tmp-d3111b3cabc0fa7b/lib/python2.7/site-packages/sunspec/core/test/test_device.py", line 402, in test_device_from_pics
    d1.from_pics(filename='pics_test_device_1.xml', pathlist=pathlist)
  File "/home/bruno/.virtualenvs/tmp-d3111b3cabc0fa7b/local/lib/python2.7/site-packages/sunspec/core/device.py", line 111, in from_pics
    raise SunSpecError('Error loading PICS: %s' % str(e))
```

bobfox, 2016-03-22:
The github generated source zip file does not contain the SunSpec 
information models as they are contained in another repository. If you 
installed from the github generated source zip file that may be the issue.

Try downloading the pysunspec-1.0.7.zip from the release and using 
setup.py to install.

bbinet, 2016-03-22:
I confirm that tests pass when using the release zip.
Thanks.

closed 2016-03-22 by bbinet