sunspec/pysunspec #9: Can't set value to rw field for SMA Sunny Island device

bbinet, 2016-03-23:
I can successfully read values from a SMA Sunny Island device, but I got the following error when I try to set a value in a rw field:
``` python
$ python
Python 2.7.9 (default, Mar  1 2015, 12:57:24) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import sunspec.core.client as client
>>> d = client.SunSpecClientDevice(client.TCP, 126, ipaddr='192.168.1.77')
>>> d.models
['common', 'model_11', 'model_12', 'inverter', 'nameplate', 'settings', 'status', 'controls', 'storage', 'volt_var', 'freq_watt_param', 'reactive_current', 'watt_pf', 'volt_watt', 'mppt']
>>> d.storage.WChaMax
>>> d.storage.read()
>>> d.storage.WChaMax
6000
>>> d.storage.WChaMax = 5900
>>> d.storage.write()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/bruno/.virtualenvs/pysunspec/local/lib/python2.7/site-packages/sunspec/core/client.py", line 303, in write
    self.model.write_points()
  File "/home/bruno/.virtualenvs/pysunspec/local/lib/python2.7/site-packages/sunspec/core/client.py", line 242, in write_points
    point.write()
  File "/home/bruno/.virtualenvs/pysunspec/local/lib/python2.7/site-packages/sunspec/core/client.py", line 259, in write
    self.block.model.device.write(int(self.addr), data)
  File "/home/bruno/.virtualenvs/pysunspec/local/lib/python2.7/site-packages/sunspec/core/client.py", line 80, in write
    raise SunSpecClientError('Modbus write error: %s' % str(e))
sunspec.core.client.SunSpecClientError: Modbus write error: Modbus exception: 1
```
According to SMA SunSpec documentation, WChaMax should be writable:

| Modbus register number | SunSpec-Name | Description / Number code(s)                                                         | Type   | Access | 
|------------------------|--------------|--------------------------------------------------------------------------------------|--------|--------| 
| 40372                  | WChaMax      | Setpoint for maximum active charging power.  Reference value for WChaGra, WDisChaGra | uint16 | RW     | 

I also tried with other rw fields, and I also got an error.

Any idea of what is wrong ?

bobfox, 2016-03-23:
Typically to write values to the SMA device, you must first apply the 
SMA GridGuard code through the Modbus interface as outlined in the SMA 
Modbus interface documentation. Consult the latest SMA Modbus interface 
document for the device you are accessing for details.

bbinet, 2016-03-24:
I had a look to SMA Modbus interface document, and it seems the 40372 rw register should not be protected by SMA GridGuard code.

Is there a way to get more debugging information on what is wrong, for example, what means error "Modbus exception: 1"?

bbinet, 2016-03-24:
I've just discovered that we can pass a `trace` function to get more debugging information, so I did:
```
>>> def tprint(s):print(s)
>>> 
>>> d = client.SunSpecClientDevice(client.TCP, 126, ipaddr='192.168.1.77', trace=tprint)
192.168.1.77:502:126 ->0000000000067E039C400003
192.168.1.77:502:126 <--0000000000097E030653756E530001
192.168.1.77:502:126 ->0000000000067E039C430001
192.168.1.77:502:126 <--0000000000057E03020042
192.168.1.77:502:126 ->0000000000067E039C860001
192.168.1.77:502:126 <--0000000000057E0302000B
192.168.1.77:502:126 ->0000000000067E039C870001
192.168.1.77:502:126 <--0000000000057E0302000D
192.168.1.77:502:126 ->0000000000067E039C950001
192.168.1.77:502:126 <--0000000000057E0302000C
192.168.1.77:502:126 ->0000000000067E039C960001
192.168.1.77:502:126 <--0000000000057E03020062
192.168.1.77:502:126 ->0000000000067E039CF90001
192.168.1.77:502:126 <--0000000000057E03020067
192.168.1.77:502:126 ->0000000000067E039CFA0001
192.168.1.77:502:126 <--0000000000057E03020032
192.168.1.77:502:126 ->0000000000067E039D2D0001
192.168.1.77:502:126 <--0000000000057E03020078
192.168.1.77:502:126 ->0000000000067E039D2E0001
192.168.1.77:502:126 <--0000000000057E0302001A
192.168.1.77:502:126 ->0000000000067E039D490001
192.168.1.77:502:126 <--0000000000057E03020079
192.168.1.77:502:126 ->0000000000067E039D4A0001
192.168.1.77:502:126 <--0000000000057E0302001E
192.168.1.77:502:126 ->0000000000067E039D690001
192.168.1.77:502:126 <--0000000000057E0302007A
192.168.1.77:502:126 ->0000000000067E039D6A0001
192.168.1.77:502:126 <--0000000000057E0302002C
192.168.1.77:502:126 ->0000000000067E039D970001
192.168.1.77:502:126 <--0000000000057E0302007B
192.168.1.77:502:126 ->0000000000067E039D980001
192.168.1.77:502:126 <--0000000000057E03020018
192.168.1.77:502:126 ->0000000000067E039DB10001
192.168.1.77:502:126 <--0000000000057E0302007C
192.168.1.77:502:126 ->0000000000067E039DB20001
192.168.1.77:502:126 <--0000000000057E03020018
192.168.1.77:502:126 ->0000000000067E039DCB0001
192.168.1.77:502:126 <--0000000000057E0302007E
192.168.1.77:502:126 ->0000000000067E039DCC0001
192.168.1.77:502:126 <--0000000000057E03020040
192.168.1.77:502:126 ->0000000000067E039E0D0001
192.168.1.77:502:126 <--0000000000057E0302007F
192.168.1.77:502:126 ->0000000000067E039E0E0001
192.168.1.77:502:126 <--0000000000057E0302000A
192.168.1.77:502:126 ->0000000000067E039E190001
192.168.1.77:502:126 <--0000000000057E03020080
192.168.1.77:502:126 ->0000000000067E039E1A0001
192.168.1.77:502:126 <--0000000000057E0302000E
192.168.1.77:502:126 ->0000000000067E039E290001
192.168.1.77:502:126 <--0000000000057E03020083
192.168.1.77:502:126 ->0000000000067E039E2A0001
192.168.1.77:502:126 <--0000000000057E03020040
192.168.1.77:502:126 ->0000000000067E039E6B0001
192.168.1.77:502:126 <--0000000000057E03020084
192.168.1.77:502:126 ->0000000000067E039E6C0001
192.168.1.77:502:126 <--0000000000057E03020040
192.168.1.77:502:126 ->0000000000067E039EAD0001
192.168.1.77:502:126 <--0000000000057E030200A0
192.168.1.77:502:126 ->0000000000067E039EAE0001
192.168.1.77:502:126 <--0000000000057E03020030
192.168.1.77:502:126 ->0000000000067E039EDF0001
192.168.1.77:502:126 <--0000000000057E0302FFFF
>>> d.storage.read()
192.168.1.77:502:126 ->0000000000067E039DB30018
192.168.1.77:502:126 <--0000000000337E03301770FFFFFFFF0000FFFFFFFF004AFFFF1496FFFF80008000FFFFFFFFFFFFFFFF000080008000800000008000FFFE8000
>>> d.storage.WChaMax
6000
>>> d.storage.WChaMax = 5900
>>> d.storage.write()
192.168.1.77:502:126 ->0000000000097E109DB3000102170C
192.168.1.77:502:126 <--0000000000037E9001
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/bruno/.virtualenvs/pysunspec/local/lib/python2.7/site-packages/sunspec/core/client.py", line 303, in write
    self.model.write_points()
  File "/home/bruno/.virtualenvs/pysunspec/local/lib/python2.7/site-packages/sunspec/core/client.py", line 242, in write_points
    point.write()
  File "/home/bruno/.virtualenvs/pysunspec/local/lib/python2.7/site-packages/sunspec/core/client.py", line 259, in write
    self.block.model.device.write(int(self.addr), data)
  File "/home/bruno/.virtualenvs/pysunspec/local/lib/python2.7/site-packages/sunspec/core/client.py", line 80, in write
    raise SunSpecClientError('Modbus write error: %s' % str(e))
sunspec.core.client.SunSpecClientError: Modbus write error: Modbus exception: 1
```
So it seems SMA device has replied with data `0000000000037E9001`: do you know what does it mean?

bobfox, 2016-03-24:
|0000000000037E9001 is a Modbus exception 1 for device id 126 (0x7E) 
operation 16 (0x10). |

bbinet, 2016-03-25:
Thanks.

In SMA SunSpec Modbus interface document, it is said that modbus exception 1 occurs when trying to write a data block to an address range that contains read-only registers.
So I decided to check that pysunspec wrote data in the correct 40372 rw modbus register by patching the trace_func code (see: #10) and it appears that pysunspec is trying to write to the 40371 ro modbus register instead...
I also tried to update `d.storage.StorCtl_Mod` value which modbus address should be 40375 (according to SMA SunSpec modbus doc), but I get the same write error (modbus exception 1), and pysunspec is still trying to write to the 40371 ro modbus register instead of the 40375 rw modbus register.

FYI, 40371 address is `Number of the following Modbus registers in accordance with the PICS table (L): 24`:

| Modbus register number | SunSpec-Name | Description / Number code(s)                                                         | Type   | Access | 
|------------------------|--------------|--------------------------------------------------------------------------------------|--------|--------| 
| 40371      |             |  Number of the following Modbus registers in accordance with the PICS table (L): 24  | uint16 | RO | 
| 40372                  | WChaMax      | Setpoint for maximum active charging power.  Reference value for WChaGra, WDisChaGra | uint16 | RW     | 
| 40375 | StorCtl_Mod | Activate hold/discharge/charge storage control mode. Bitfield value: 0 | uint16 | RW | 

Any idea why pysunspec is always trying to write to the 40371 ro register?

bbinet, 2016-03-25:
I think I now understand the current behaviour: when I do `d.storage.write()`, pysunspec will try to write the whole storage address block range, which contains some read-only blocks (40371 address is read-only).
So I think that pysunspec should only try to write what has actually been changed, isn't it?

So in my case, if I only update `d.storage.WChaMax` and `d.storage.StorCtl_Mod` values, it should not try to write the whole block range (which might contain some read-only block), but only these 2 values.

What do you think about it?

bbinet, 2016-03-25:
Or is there already a way to write single values and not a whole block of values?

bobfox, 2016-03-25:
The pysunspec library should only be writing to registers that have been 
changed. Can you send the trace of modifying a point and the subsequent 
Modbus activity when performing the write?

bbinet, 2016-03-25:
You want something different than this previous comment: https://github.com/sunspec/pysunspec/issues/9#issuecomment-200739400 ?

bobfox, 2016-03-25:
> You want something different than this previous comment: #9 (comment)

For the case in the previous comment, it appears that the SMA 
documentation is not showing the address on the line but the line 
address + 1 which is common for Modbus notation. If you look at the 
previous read, the value coming from line address 40371 is 0x1770 or 
6000. This indicates to me that WChaMax is probably at line address 
40371 so the write address is correct.

bbinet, 2016-03-29:
Ok, you are right, it must show address + 1 if it is common for Modbus notation: there is no error when doing the same pysunspec trace when trying to modify `StorCtl_Mod` value (address 40375):

``` python
$ python
Python 2.7.9 (default, Mar  1 2015, 12:57:24) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import sunspec.core.client as client
>>> d = client.SunSpecClientDevice(client.TCP, 126, ipaddr='192.168.1.77')
>>> d.storage.read()
192.168.1.77:502:126[addr=40371] ->0000000000067E039DB30018
192.168.1.77:502:126[addr=40371] <--0000000000337E03301770FFFFFFFF0000FFFFFFFF0047FFFF1374FFFF80008000FFFFFFFFFFFFFFFF000080008000800000008000FFFE8000
>>> d.storage.StorCtl_Mod
0
>>> d.storage.StorCtl_Mod = 1
>>> d.storage.write()
192.168.1.77:502:126[addr=40374] ->0000000000097E109DB60001020001
192.168.1.77:502:126[addr=40374] <--0000000000067E109DB60001
```

But when I read the same value again, it seems it has not been taken into account:
```
>>> d.storage.read()
192.168.1.77:502:126[addr=40371] ->0000000000067E039DB30018
192.168.1.77:502:126[addr=40371] <--0000000000337E03301770FFFFFFFF0000FFFFFFFF0047FFFF1374FFFF80008000FFFFFFFFFFFFFFFF000080008000800000008000FFFE8000
>>> d.storage.StorCtl_Mod
0
```
I guess this is not due to pysunspec but to my SMA Sunny Island device instead. If you still have any idea why it did not persist, don't hesitate to explain me in a comment.

Regarding the `WChaMax` write issue, I don't know why it does not work, but it seems to be due to this SMA device too. I will try to get answers from SMA support.

Thanks for your help!

closed 2016-04-07 by bobfox

bbinet, 2017-04-24:
No, I gave up with pysunspec to write values to the Sunny Island as it does not seem to be mature enough and I've ended using the SMA modbus protocol directly.

smeop, 2017-04-24:
Thx a lot! SMA support suggested that 2 hours ago, too.