sunspec/pysunspec #16: Python 3 support

altendky, 2016-08-15:
Is there any interest in a pull request providing both Python 2 and Python 3 support?

altendky, 2016-08-17:
I haven't used it beyond verifying that `pysunspec_test.py` passes, but it works at least that well in both 2 and 3.

https://github.com/altendky/pysunspec/tree/a795be88bb9ac4f6db6dd48752ec867e807a205f

I'll let it settle a bit and put it to use before filing a formal pull request.

bobfox, 2016-08-17:
If you have time, I would like to talk to you about this in more detail. I can be reached at bob@sunspec.org. Thanks, Bob Fox

altendky, 2016-08-19:
Hi Bob,  I emailed you Wednesday and am happy to chat.  I just wanted to make sure my email didn't get lost.

altendky, 2017-06-30:
I briefly reviewed the differences in py2/3 support approaches between my branch and @jbm950's.  Along with a few inline comments in #24 I noticed the following general points regarding approach.

* You avoided using __future__ imports which means that code like `print(...)` will have different effects in different versions and also different features.  Similar for `/` (division).
* You avoided a 2/3 compatibility library which is nice but I'm not sure it really pans out when trying to write compatible code.  Perhaps the `sys.version_info` checks are sufficient but I'm skeptical.

Perhaps some of this relates to what versions of py2 must be supported but in general I think it's valuable to handle as much as we can without if/else version checks.  Since either branch/approach introduces many changes it seems like getting the requisite py2 platforms into testing before updating to py3 would be best (#44).