sunspec/pysunspec #17: Tests are failing with an exception

altendky, 2016-08-15:
Are the tests passing for other people?  They fail with an exception for me.  I'll work this out but figured it would be worth checking here.

```python
/epc/t/313/pysunspec/venv/bin/python scripts/pysunspec_test.py
pySunSpec version: 1.0.7
Test device path: /epc/t/313/pysunspec/venv/lib/python2.7/site-packages/sunspec/core/test/devices

Traceback (most recent call last):
  File "scripts/pysunspec_test.py", line 13, in <module>
    test.test_all()
  File "/epc/t/313/pysunspec/venv/lib/python2.7/site-packages/sunspec/core/test/test_all.py", line 58, in test_all
    (count_run, count_passed, count_failed) = module.test_all(pathlist, stop_on_failure)
  File "/epc/t/313/pysunspec/venv/lib/python2.7/site-packages/sunspec/core/test/test_device.py", line 577, in test_all
    if test(pathlist) is True:
  File "/epc/t/313/pysunspec/venv/lib/python2.7/site-packages/sunspec/core/test/test_device.py", line 418, in test_device_from_pics
    d1.from_pics(filename='pics_test_device_1.xml', pathlist=pathlist)
  File "/epc/t/313/pysunspec/venv/lib/python2.7/site-packages/sunspec/core/device.py", line 127, in from_pics
    raise SunSpecError('Error loading PICS: %s' % str(e))
sunspec.core.util.SunSpecError: Error loading PICS: Block index out of range: 1
```

Or, with an `except` removed to see the original trace:

```python
/epc/t/313/pysunspec/venv/bin/python scripts/pysunspec_test.py
pySunSpec version: 1.0.7
Test device path: /epc/t/313/pysunspec/venv/lib/python2.7/site-packages/sunspec/core/test/devices

Traceback (most recent call last):
  File "scripts/pysunspec_test.py", line 13, in <module>
    test.test_all()
  File "/epc/t/313/pysunspec/venv/lib/python2.7/site-packages/sunspec/core/test/test_all.py", line 58, in test_all
    (count_run, count_passed, count_failed) = module.test_all(pathlist, stop_on_failure)
  File "/epc/t/313/pysunspec/venv/lib/python2.7/site-packages/sunspec/core/test/test_device.py", line 577, in test_all
    if test(pathlist) is True:
  File "/epc/t/313/pysunspec/venv/lib/python2.7/site-packages/sunspec/core/test/test_device.py", line 418, in test_device_from_pics
    d1.from_pics(filename='pics_test_device_1.xml', pathlist=pathlist)
  File "/epc/t/313/pysunspec/venv/lib/python2.7/site-packages/sunspec/core/device.py", line 120, in from_pics
    model.from_pics(m)
  File "/epc/t/313/pysunspec/venv/lib/python2.7/site-packages/sunspec/core/device.py", line 481, in from_pics
    raise SunSpecError('Block index out of range: %s' % (str(block_index)))
sunspec.core.util.SunSpecError: Block index out of range: 1
```

altendky, 2016-08-15:
You must checkout the sunspec/models repository then things work better.  In my fork I added it as a submodule (7e483f9bc6fb23d1e51bb4f043df4b707b138db5).

closed 2016-08-15 by altendky

tank-2, 2016-09-28:
This error is not at all obvious, I just spent too much time tracking this down on my own, until thinking to take a look at the closed issues. I discovered the fix only after reverting to 1.0.6 and noticing the models directory. Why not just include the models directory in 1.0.7 to save people like me trouble?

bobfox, 2016-10-03:
I am sorry this caused you to waste some time. The models are included with every release. The pysunspec-1.07.zip is the 1..0.7 release in the standard Python format. If you install the release in the standard way, the models are included. If you pull the source code directly, you will only get the source code. The models are in a separate repository because they are used in other places and are independent of any on the Python projects.

altendky, 2016-10-04:
As in 7e483f9, the models could be included as Git submodules.  That would at least use the 'standard' Git tool for linking repositories.  That said, many people would still likely miss it (myself included).  Seems to me that it would be reasonable to have an explicit check for that directory which raises an exception when it is not found and provides both a direct message as to what is wrong ('Models are not available in standard location') as well as instructions to correct the issue ('Checkout http://github.com/sunspec/models into the ... directory').  Or, Git submodules instructions, if that is how they will be acquired.  Libraries and applications also often allow for environment variables to specify non-standard locations for 'external' resources.

Another approach focusing on the Python side would be to integrate fetching of the modules into `setup.py`.  Though, the checks and error reporting listed above would likely still be applicable.

If models isn't going to become a Git submodule I would recommend it exist outside of the checked-out pysunspec Git repository rather than as a subdirectory.  I personally find nested unrelated repositories confusing.