sunspec/pysunspec #39: Improving error reporting by masking less

altendky, 2017-06-14:
Masking exceptions often makes it harder to understand and fix the problem, not easier.  For example, running unit tests without having the models cloned (such as in #35 and others) results in this unhelpful message.

```
======================================================================
ERROR: test_client_device (sunspec.core.test.test_client.TestClientDevice)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/epc/t/612/pysunspec/sunspec/core/test/test_client.py", line 48, in test_client_device
    pathlist=self.pathlist)
  File "/epc/t/612/pysunspec/sunspec/core/device.py", line 175, in from_pics
    raise SunSpecError('Error loading PICS: %s' % str(e))
SunSpecError: Error loading PICS: Block index out of range: 1
```

That is masking a slightly better custom message which would have read:

```
Model file for model model 1 not found
```

which is masking the actually useful (and automatic) message:

```
[Errno 2] No such file or directory: '/epc/t/612/pysunspec/sunspec/core/../models/smdx/smdx_00001.xml'
```

Now certainly there are times that useful information can be added to a standard message.  Above could be enhanced by a mention of the `models` subrepo and the search path feature so that a user/developer can look at the docs related to those topics.

In Python 3 you can simply `raise ... from ...` as below but I have not yet found a clean 2/3 compatible way to get a result like this.

```
altendky@tp:~$ cat rf.py
try:
    raise Exception('inner')
except Exception as e:
    raise Exception('outer') from e
```
```
altendky@tp:~$ python3 rf.py
Traceback (most recent call last):
  File "rf.py", line 2, in <module>
    raise Exception('inner')
Exception: inner

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "rf.py", line 4, in <module>
    raise Exception('outer') from e
Exception: outer
```

Regardless of chaining, is there any interest in discussing improving the clarity of the errors?

jbm950, 2017-06-30:
Yes there is interest. I know when trying to update the repository to python 3 a few months ago I had to comment out some of the try except statements to find the actual error to fix and so I believe this would be really useful. The repository is trying to retain python 2/3 status though and so while the chaining looks very useful it would have to work in both versions.

altendky, 2017-06-30:
I feel this is a big deal but I think getting the branches back together comes first. On the chaining, yeah, I was disappointed not to find a 2/3 option so that makes it harder to add helpful messages.