sunspec/pysunspec #44: Setup testing for all supported Python versions

altendky, 2017-06-30:
I tend towards keeping up to date so the reference to 2.4 (replaced by 2.5 over a decade ago) in the readme is a bit scary.  But, whatever versions we intend to support should be included in testing.  It looks like [Travis only provides 2.7 and 2.6](https://docs.travis-ci.com/user/languages/python/#Choosing-Python-versions-to-test-against) (in the py2 family) so we may need to do something special (install old versions from a ppa?) or add another testing platform to get coverage.  [tox](https://tox.readthedocs.io/en/latest/) may also be a good piece of the puzzle.

So, is the 2.4-2.7 reference in the readme the official list of what we need to support?  Latest patch level on each?  Obviously with interest in py3 versions in the future.

jbm950, 2017-07-07:
Sorry for the delay, the holiday weekend disrupted my productivity. In talking with Bob on creating a py2/3 version he was ok with moving towards supporting 2.7 from the python 2 family because some of the cross compatibility features were introduced in this version and so this would allow easier maintenance. We would just reference people to previous releases for older versions of python I think.

altendky, 2017-07-07:
Not a problem.  When I can respond quickly I do, but I understand that we get pulled different directions at various times.  I took a 8 months or whatever off this after I first made it work.

Anyways, the readme should get updated but perhaps we address my rewrite first rather than mixing more stuff in there.  Who knows, perhaps I'll need to rewrite my rewrite.  :]  I think that 2.6 has most of the compatibility stuff as well so it is probably doable, but I'm certainly not going to fight for that.  In py3 I'm not sure exactly on the details but I think that 3.4+ is a common cutoff, but if it works in 3.3 (which you include in the tests) then hey, leave it for now I suppose.

We should probably add 3.6, I wouldn't expect there to be any issues and Travis has it.  Might be worth adding 3.7-dev as well (and keeping it an optional failure) just to catch issues sooner since Travis has it as well.

So, plan to claim and require compatibility with 2.7 now and 3.4-3.6 with the py3 support?  I'd add 2.6 to the tests and see how it goes and save for 3.3 and 3.7-dev.  I can follow plan to follow up with PRs on both the test and readme.