sunspec/pysunspec #52: suns.py does not work on a solaredge inverter

bbinet, 2017-10-20:
Here is the traceback when I try to get sunspec data out of our solaredge inverter with the `suns.py` command:
```
$ suns.py -i 192.168.2.26 -a 1 -T 30

Timestamp: 2017-10-20T15:29:41Z
Traceback (most recent call last):
  File "/usr/bin/suns.py", line 88, in <module>
    sd.read()
  File "/usr/lib/python2.7/dist-packages/sunspec/core/client.py", line 477, in read
    self.device.read_points()
  File "/usr/lib/python2.7/dist-packages/sunspec/core/client.py", line 101, in read_points
    model.read_points()
  File "/usr/lib/python2.7/dist-packages/sunspec/core/client.py", line 246, in read_points
    raise SunSpecClientError(e)
sunspec.core.client.SunSpecClientError: Modbus read error: Connection error: [Errno 111] Connection refused
```

Note that the inverter modbus seems to work though as I'm able to retrieve data through [a perl tcp modbus client](https://github.com/sourceperl/mbtget):
```
$ perl mbtget -r3 -n 100 -a 40001 192.168.2.26
values:
  1 (ad 40001): 28243
  2 (ad 40002):     1
  3 (ad 40003):    65
  4 (ad 40004): 21359
  5 (ad 40005): 27745
  6 (ad 40006): 29253
  7 (ad 40007): 25703
  8 (ad 40008): 25888
  9 (ad 40009):     0
 10 (ad 40010):     0
 11 (ad 40011):     0
 12 (ad 40012):     0
 13 (ad 40013):     0
 14 (ad 40014):     0
 15 (ad 40015):     0
 16 (ad 40016):     0
 17 (ad 40017):     0
 18 (ad 40018):     0
 19 (ad 40019):     0
 20 (ad 40020): 21317
 21 (ad 40021): 13110
 22 (ad 40022): 14384
 23 (ad 40023):     0
 24 (ad 40024):     0
 25 (ad 40025):     0
 26 (ad 40026):     0
 27 (ad 40027):     0
 28 (ad 40028):     0
 29 (ad 40029):     0
 30 (ad 40030):     0
 31 (ad 40031):     0
 32 (ad 40032):     0
 33 (ad 40033):     0
 34 (ad 40034):     0
 35 (ad 40035):     0
 36 (ad 40036):     0
 37 (ad 40037):     0
 38 (ad 40038):     0
 39 (ad 40039):     0
 40 (ad 40040):     0
 41 (ad 40041):     0
 42 (ad 40042):     0
 43 (ad 40043):     0
 44 (ad 40044): 12336
 45 (ad 40045): 12339
 46 (ad 40046): 11826
 47 (ad 40047): 12337
 48 (ad 40048): 13824
 49 (ad 40049):     0
 50 (ad 40050):     0
 51 (ad 40051):     0
 52 (ad 40052): 14131
 53 (ad 40053): 12593
 54 (ad 40054): 12343
 55 (ad 40055): 13123
 56 (ad 40056):     0
 57 (ad 40057):     0
 58 (ad 40058):     0
 59 (ad 40059):     0
 60 (ad 40060):     0
 61 (ad 40061):     0
 62 (ad 40062):     0
 63 (ad 40063):     0
 64 (ad 40064):     0
 65 (ad 40065):     0
 66 (ad 40066):     0
 67 (ad 40067):     0
 68 (ad 40068):     2
 69 (ad 40069):   101
 70 (ad 40070):    50
 71 (ad 40071):   102
 72 (ad 40072):   102
 73 (ad 40073): 65535
 74 (ad 40074): 65535
 75 (ad 40075): 65534
 76 (ad 40076):  2374
 77 (ad 40077): 65535
 78 (ad 40078): 65535
 79 (ad 40079): 65535
 80 (ad 40080): 65535
 81 (ad 40081): 65535
 82 (ad 40082): 65535
 83 (ad 40083): 15959
 84 (ad 40084): 65534
 85 (ad 40085):  5012
 86 (ad 40086): 65534
 87 (ad 40087): 24435
 88 (ad 40088): 65534
 89 (ad 40089): 18503
 90 (ad 40090): 65534
 91 (ad 40091):  6531
 92 (ad 40092): 65534
 93 (ad 40093):    30
 94 (ad 40094):  7612
 95 (ad 40095):     0
 96 (ad 40096):  4280
 97 (ad 40097): 65532
 98 (ad 40098):  3785
 99 (ad 40099): 65535
100 (ad 40100): 16202
```

Any idea what is wrong with `suns.py` here?

mcmellow, 2018-01-24:
I have a similar problem as bbinet.  See log below.  Is there an alternative? Modbus status is ready at the panal of the Solaredge Inverter

pi@mcmellow-pi:~/pysunspec $ suns.py -i 192.168.xxx.yy  -P 502 -a 1 -T 60

Timestamp: 2018-01-24T15:41:28Z
Traceback (most recent call last):
  File "/usr/local/bin/suns.py", line 88, in <module>
    sd.read()
  File "/usr/local/lib/python2.7/dist-packages/sunspec/core/client.py", line 813, in read
    self.device.read_points()
  File "/usr/local/lib/python2.7/dist-packages/sunspec/core/client.py", line 204, in read_points
    model.read_points()
  File "/usr/local/lib/python2.7/dist-packages/sunspec/core/client.py", line 382, in read_points
    raise SunSpecClientError(e)
sunspec.core.client.SunSpecClientError: Modbus read error: Connection error: [Errno 111] Connection refused

Heiner92, 2018-06-05:
I have encountered similar problems with older inverters (SolarMax) using pysunspec.
I'm guessing these inverters simply don't support the sunspec modbus standard.
After some research I found this python lib for solarmax communication (it is the same as mentioned by @bbinet - but in python).

[python-solarmax](https://github.com/bwurst/python-solarmax)

All tested with python 2.7 and working fine.
I have created a python 3 branch which still needs some testing. 
Hope this helps you.

chriskelman, 2018-08-10:
Have you tried a different device ID? For example it seems SMA uses ID: 126 rather than Id: 1

fgnn, 2019-11-11:
I came across this exact same error on a brand new solaredge inverter:
SE6000H-US000BNU4 with firmware 4.7.19

Looking at the wireshark logs, I found that the code does a series of 9 modbus reads before closing the socket. The error occurs when the code tried to open a new socket 40ms later. The inverter responds with the RST flag. Maybe the inverter only allows a single connection, and we are trying to open a new socket before it can fully clean up after the last one?

As a workaround, I added the following line at line 333 in core.py to introduce a delay. 
`time.sleep(0.1) `
This seems to work 99% of the time now. I now see that it opens and closes the socket 5 times and took about 990ms to read all values.

A better long term solution would be to perform all modbus reads with the same socket to avoid this situation entirely.

marpi82, 2019-11-30:
> I came across this exact same error on a brand new solaredge inverter:
> SE6000H-US000BNU4 with firmware 4.7.19
>
> Looking at the wireshark logs, I found that the code does a series of 9 modbus reads [...]

I have the same issue, adding ```time.sleep(0.1)``` in pysunspec/sunspec/core/client.py line 333 helps

altendky, 2019-11-30:
Hopefully manufacturer specific issues are being pointed out to the manufacturers.  Even if it is determined that there is an issue with pysunspec maybe they could help out with a PR to fix it.

marpi82, 2019-11-30:
> Hopefully manufacturer specific issues are being pointed out to the manufacturers. Even if it is determined that there is an issue with pysunspec maybe they could help out with a PR to fix it.

I have already sent several service requests to SE.  One of them was that the phase-to-phase voltage of the meter connected to the inverter and read from the inverter is roughly -230V, when I read it directly from the meter it is about 400V.  For today, the request has not been processed.  Posted 6 months ago.  So I wouldn't count on SolarEdge on this issue :)

MojaveTom, 2020-04-30:
As of April 2020, SolarEdge has not fixed this problem.

Model:  SE7600H-US000BNU4
Communications CPU:     4.8.28

I added code in /sunspec/core/modbus/client.py to ensure at least a 50ms delay between disconnect and connect.

_Not thoroughly tested yet._

I think that the pysunspec library is the appropriate place to fix this.  Manufacturers may have hardware limitations on the speed at which connections can be re-made.  Wouldn't it be nice if this was a parameter in the "common" block?

As @fgnn mentions, the sunspec library is connecting and disconnecting multiple times to read data; his suggestion of using the same connection for multiple operations has merit.  When doing a device.read(), the library connects, reads the first module's data, disconnects, then immediately reconnects to read the next module's data.  This is because the modbus.client.read() function checks to see if a connection is open, and if not opens one for just this read operation, then disconnects.  When doing a device.read(), that code should handle the connect/disconnect instead of pushing it to the modbus layer.

pkgdata, 2020-05-10:
Hi,

I tried to locate and write the code "time.sleep(0.1)" but get still the same result as bbinet: (In both pysunspec/sunspec/core/client.py and pysunspec/sunspec/core/modbus/client.py

pi@solaredgepi:~/pysunspec $ suns.py -i 192.168.1.186 -P 1502 -a 1 -T 60
Traceback (most recent call last):
  File "/usr/local/bin/suns.py", line 71, in <module>
    sd = client.SunSpecClientDevice(client.TCP, options.a, ipaddr=options.i, ipport=options.P, timeout=options.T)
  File "/usr/local/lib/python2.7/dist-packages/sunspec/core/client.py", line 775, in __init__
    self.device.scan(progress=scan_progress, delay=scan_delay)
  File "/usr/local/lib/python2.7/dist-packages/sunspec/core/client.py", line 218, in scan
    self.modbus_device.connect()
  File "/usr/local/lib/python2.7/dist-packages/sunspec/core/modbus/client.py", line 678, in connect
    raise ModbusClientError('Connection error: %s' % str(e))
sunspec.core.modbus.client.ModbusClientError: Connection error: [Errno 111] Connection refused

The inverter can communicate to an another pi, (home assistans). I dont know how I managed that... so the inverter configured to communicate.

MojaveTom, 2020-05-11:
@pkgdata I haven't researched thoroughly, but I suspect the sunspec library is trying to open the client twice.  The SolarEdge documentation specifically states that the device only supports ONE connection at a time.  I think (again more research needed) that the reason putting in a delay works is that it gives time for the device to complete and disconnect before the next connection begins.  I encountered the problem if I used `device.read()`, but avoided the problem if I read each module individually.

I'm not ready to publish my work-around, but had to use class variables to ensure an adequate delay; instance variables didn't work.  (Thus the belief that sunspec is opening multiple instances of the client.)

I don't believe the problem is with SolarEdge.  Your other Pi may be slow enough or using the library in a way that doesn't encounter the problem.

pkgdata, 2020-05-11:
@MojaveTom , thank you for answer...

As some have written before, it seems to work when making a time.sleep (0.1)
But I'm not a programmer and when I type this line in line 333 doesn't seem to have helped. (in addition, the 2 client.py files are two completely different files, and line 333 looks different)

in /home/pi/pysunspec/sunspec/core/client.py
    def read_points(self):
        """Read all points in the model from the physical device.
        """

        if self.model_type is not None:
            # read current model
            try:
                end_index = len(self.read_blocks)
                Row 333: ........... if end_index == 1:
                    data = self.device.read(self.addr, self.len)
                else:
                    data = b''
                    index = 0
                    while index < end_index:
                        addr = self.read_blocks[index]
                        index += 1
                        if index < end_index:
                            read_len = self.read_blocks[index] - addr
                        else:
                            read_len = self.addr + self.len - addr
                        data += self.device.read(addr, read_len)
                if data:
                    # print('data len = ', len(data))
                    data_len = len(data)/2
                    if data_len != self.len:
                        raise SunSpecClientError('Error reading model %s' % self.model_type)
----------------------------------------------------------------------------------------------------
in /home/pi/pysunspec/sunspec/core/modbus/client.py
                else:
                    return
        else:
            raise ModbusClientError('Client serial port not open: %s' % self.name)

        if sys.version_info > (3,):
            resp = bytes(resp, 'latin-1')

        return resp
	row 333.......
    def _write(self, slave_id, addr, data, trace_func=None):
        resp = ''
        len_remaining = 5
        len_found = False
        except_code = None
        func = FUNC_WRITE_MULTIPLE
        len_data = len(data)
        count = int(len_data/2)

Are these the right lines?
Should I put time.sleep (0.1) in both?

MojaveTom, 2020-05-11:
@pkgdata Putting a delay in `/pysunspec/sunspec/core/modbus/client.py` is the more effective place; this is the lowest level module to actually interact with the TCP connection.  Putting delays at a specific line number is risky since as you noticed your sources may be different.  I'm sure you just want to get your project going, as I did, so I'll attach my modified file.  Realize that this is a POOR EXCUSE for finding the real problem and fixing that.  This just ensures a .05 sec delay between socket (TCP) disconnect and connect.  As I said, this is a band-aid to let the SolarEdge inverter finish up one connection before starting another.  The real solution is to not try to make multiple connections so quickly, instead just re-use an already open connection.  I have only tested this code on a Mac Mini, I have no idea if it will work on your Pi.

The parameter that sets the delay time is called `MIN_RECONNECT_TIME_SEC` defined near the top of the file.  The code modifications are in the class `ModbusClientDeviceTCP`; functions `connect()` and `disconnect()` which are near the middle of the file.  The name of the file indicates which file from the distribution is to be replaced with the modified file (name changed to the name of the replaced file). [Looks like I can only include a zip of the file, so you have to unzip it before using it.]

Some day when I'm looking for a distraction, I'll dig into the sunspec code and fix it correctly.

Good luck with this,
Tom
[Modified-sunspec-core-modbus-client.py.zip](https://github.com/sunspec/pysunspec/files/4610344/Modified-sunspec-core-modbus-client.py.zip)