sunspec/pysunspec #53: pysunspec for python 3.6

learnetk, 2018-03-26:
hi 
i try to test the installation with Python3.6 using python -m unittest discover -v sunspec and get errors

Ran 40 tests in 0.046s
FAILED (errors=14)

Does pysunspec support Python3 already ?

Thanks

jbm950, 2018-03-26:
I believe the Python 3 branch was merged. Would you mind showing what errors you were getting?

learnetk, 2018-03-27:
i have the latest version of python (3.6) . The errors i get is as under. It would of great help if you could have a look.

C:\Users\user\Documents\python\pysunspec-master\pysunspec-master>python setup.py install
running install
running build
running build_py
running build_scripts
running install_lib
running install_scripts
running install_egg_info
Removing C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\pysunspec-2.0.0-py3.6.egg-info
Writing C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\pysunspec-2.0.0-py3.6.egg-info

C:\Users\user\Documents\python\pysunspec-master\pysunspec-master>python -m unittest discover -v sunspec
core.test.test_client (unittest.loader._FailedTest) ... ERROR
test_data (core.test.test_data.TestData) ... ok
test_device_blocktype_not_equal (core.test.test_device.TestDevice) ... ERROR
test_device_common_len_65 (core.test.test_device.TestDevice) ... ERROR
test_device_constant_sf (core.test.test_device.TestDevice) ... ERROR
test_device_from_pics (core.test.test_device.TestDevice) ... ERROR
test_device_models_smdx (core.test.test_device.TestDevice) ... ERROR
test_device_modeltype (core.test.test_device.TestDevice) ... ERROR
test_device_modeltype_not_equal (core.test.test_device.TestDevice) ... ERROR
test_device_pointtype (core.test.test_device.TestDevice) ... ERROR
test_device_pointtype_not_equal (core.test.test_device.TestDevice) ... ERROR
test_device_to_pics (core.test.test_device.TestDevice) ... ERROR
test_device_value_get (core.test.test_device.TestDevice) ... ERROR
test_device_value_set (core.test.test_device.TestDevice) ... ERROR
core.test.test_modbus_client (unittest.loader._FailedTest) ... ERROR
test_modbus_mbmap_from_xml_element (core.test.test_modbus_mbmap.TestModbusMap) ... C:\Users\user\Documents\python\pysunspec-master\pysunspec-master\sunspec\core\modbus\mbmap.py:172: ResourceWarning: unclosed file <_io.BufferedReader name='C:\\Users\\user\\Documents\\python\\pysunspec-master\\pysunspec-master\\sunspec\\core\\test\\devices\\mbmap_test_device_1_a.xml'>
  map_data = pathlist.read(filename)
C:\Users\user\Documents\python\pysunspec-master\pysunspec-master\sunspec\core\modbus\mbmap.py:172: ResourceWarning: unclosed file <_io.BufferedReader name='C:\\Users\\user\\Documents\\python\\pysunspec-master\\pysunspec-master\\sunspec\\core\\test\\devices\\mbmap_test_device_1_b.xml'>
  map_data = pathlist.read(filename)
C:\Users\user\Documents\python\pysunspec-master\pysunspec-master\sunspec\core\modbus\mbmap.py:172: ResourceWarning: unclosed file <_io.BufferedReader name='C:\\Users\\user\\Documents\\python\\pysunspec-master\\pysunspec-master\\sunspec\\core\\test\\devices\\mbmap_test_device_1_c.xml'>
  map_data = pathlist.read(filename)
ok
test_modbus_mbmap_from_xml_file (core.test.test_modbus_mbmap.TestModbusMap) ... C:\Users\user\Documents\python\pysunspec-master\pysunspec-master\sunspec\core\modbus\mbmap.py:172: ResourceWarning: unclosed file <_io.BufferedReader name='C:\\Users\\user\\Documents\\python\\pysunspec-master\\pysunspec-master\\sunspec\\core\\test\\devices\\mbmap_test_device_1.xml'>
  map_data = pathlist.read(filename)
ok
test_data_to_double (core.test.test_util.TestUtil) ... ok
test_data_to_eui48 (core.test.test_util.TestUtil) ... ok
test_data_to_float (core.test.test_util.TestUtil) ... ok
test_data_to_ipv6addr (core.test.test_util.TestUtil) ... ok
test_data_to_s16 (core.test.test_util.TestUtil) ... ok
test_data_to_s32 (core.test.test_util.TestUtil) ... ok
test_data_to_s64 (core.test.test_util.TestUtil) ... ok
test_data_to_str (core.test.test_util.TestUtil) ... ok
test_data_to_u16 (core.test.test_util.TestUtil) ... ok
test_data_to_u32 (core.test.test_util.TestUtil) ... ok
test_data_to_u64 (core.test.test_util.TestUtil) ... ok
test_eui48_to_data (core.test.test_util.TestUtil) ... ok
test_float32_to_data (core.test.test_util.TestUtil) ... ok
test_float_to_data (core.test.test_util.TestUtil) ... ok
test_float_to_data32 (core.test.test_util.TestUtil) ... ok
test_ipv6addr_to_data (core.test.test_util.TestUtil) ... ok
test_s16_to_data (core.test.test_util.TestUtil) ... ok
test_s32_to_data (core.test.test_util.TestUtil) ... ok
test_s64_to_data (core.test.test_util.TestUtil) ... ok
test_str_to_data (core.test.test_util.TestUtil) ... ok
test_u16_to_data (core.test.test_util.TestUtil) ... ok
test_u32_to_data (core.test.test_util.TestUtil) ... ok
test_u64_to_data (core.test.test_util.TestUtil) ... ok

======================================================================
ERROR: core.test.test_client (unittest.loader._FailedTest)
----------------------------------------------------------------------
ImportError: Failed to import test module: core.test.test_client
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\lib\unittest\loader.py", line 428, in _find_test_path
    module = self._get_module_from_name(name)
[... 232 lines cut ...]
  File "C:\Users\user\Documents\python\pysunspec-master\pysunspec-master\sunspec\core\test\test_device.py", line 306, in setUp
    device.check_for_models(pathlist=self.pathlist)
  File "C:\Users\user\Documents\python\pysunspec-master\pysunspec-master\sunspec\core\device.py", line 909, in check_for_models
    'Unable to open common model (1).  '
Exception: Unable to open common model (1).  Make certain model repository is available.  Please reference the README.md.

======================================================================
[... cut ...]
    __import__(name)
  File "C:\Users\user\Documents\python\pysunspec-master\pysunspec-master\sunspec\core\test\test_modbus_client.py", line 29, in <module>
    import sunspec.core.modbus.client as modbus
  File "C:\Users\user\Documents\python\pysunspec-master\pysunspec-master\sunspec\core\modbus\client.py", line 27, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'

----------------------------------------------------------------------
Ran 40 tests in 0.067s

FAILED (errors=14)

jbm950, 2018-03-27:
Well the good news is that none of those appear to be Python 3 issues. It looks like you're running into two issues. 

`ModuleNotFoundError: No module named 'serial'` - Looks like you don't have pyserial installed. Please refer to the pyserial section in the installation instructions in the readme

`Exception: Unable to open common model (1). Make certain model repository is available. Please reference the README.md.` - There's a sub-repository that holds the models because they're independent of pysunspec. It doesn't look like a new release has been done since the python 3 branch was merged so the best bet to get it installed correctly is by using the git commands as detailed in the pysunspec section in the installation instructions in the readme (alternatively you could manually download the folder and place it in the correct spot in pysunspec source code. The repository is another repository in the sunspec organization).

learnetk, 2018-03-28:
Thankyou so much for your support.That sounds great that its not a Python 3 issue :)

Actually to be sure that i understand the documentation right, i tried pysunspec with python 2 and was successful in testing the unittest discover command really fast. I didnt have to make any changes at all.

Python3 Issue :

i have Python 3.6 installed. As mentioned in the documentation, i downloaded and succesfully installed the lastet version of pyserial(3.4 ). For the latest version of pysunspec (Release 1.0.8) i downloaded it from https://github.com/sunspec/pysunspec/releases/
As you adviced i got the Models (smdx folder) from the Model Repositoray (https://github.com/sunspec/models) and placed it under /pysunspec-1.0.8/sunspec/models/

Now when i try to install pysunspec i get errors as shown below. Being new to Python i am sure that it could be a mistake i cannot track.

C:\Project\Python\py3\py -3 setup.py install > install.txt 2>&1

running install
running build
running build_py
running build_scripts
running install_lib
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\client.py to client.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\device.py to device.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\modbus\client.py to client.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\modbus\mbmap.py to mbmap.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\smdx.py to smdx.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\fake\serial.py to serial.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\fake\socket.py to socket.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_all.py to test_all.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_client.py to test_client.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_data.py to test_data.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_device.py to test_device.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_modbus_client.py to 

test_modbus_client.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_modbus_mbmap.py to 

test_modbus_mbmap.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\util.py to util.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\models\smdx\manifest.py to manifest.cpython-36.pyc
running install_scripts
running install_egg_info
Writing C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\pysunspec-1.0.8-py3.6.egg-info
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\client.py", line 69
    except modbus.ModbusClientError, e:
                                   ^
SyntaxError: invalid syntax

  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\device.py", line 119
    except Exception, e:
                    ^
SyntaxError: invalid syntax

  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\modbus\client.py", line 119
    except Exception, e:
                    ^
SyntaxError: invalid syntax

  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\modbus\mbmap.py", line 108
    except Exception, e:
                    ^
SyntaxError: invalid syntax

  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\smdx.py", line 129
    except Exception, e:
                    ^
SyntaxError: invalid syntax

Sorry: TabError: inconsistent use of tabs and spaces in indentation (serial.py, line 62)
Sorry: TabError: inconsistent use of tabs and spaces in indentation (socket.py, line 51)
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_all.py", line 49
    print 'Adding local path: ', local_path
                              ^
[... 11 lines cut ...]

  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_device.py", line 307
    except Exception, e:
                    ^
SyntaxError: invalid syntax

  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_modbus_client.py", line 31
    print s
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(int s)?

  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_modbus_mbmap.py", line 60
    except Exception, e:
                    ^
SyntaxError: invalid syntax

  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\util.py", line 226
    except Exception, e:
                    ^
SyntaxError: invalid syntax

  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\models\smdx\manifest.py", line 89
    except Exception, e:
                    ^
SyntaxError: invalid syntax

Heiner92, 2018-03-28:
All lines handling exceptions need to be changend for usage with python 3
`except Exception, e:`   
should be  
`except Exception as e:`

learnetk, 2018-03-28:
Hi Heiner,
Thanks for pointing that out . Then it seems that pysunspec may not work for python3 as i see many print statements as for python 2

Thanks

jbm950, 2018-03-28:
As I mentioned in my last comment `It doesn't look like a new release has been done since the python 3 branch was merged` (the last release was Feb 2017 and the Python 3 branch was merged ~Febuary **2018**). This means you'll either have to use the git commands as detailed in the readme or download the .zip from the master branch and manually move the models folder like you did last time.

learnetk, 2018-03-28:
Thankyou for helping me out with this. i did the following steps. please correct me if something was wrong

Step 1 : download (sunspec/pysunspec) zip file from the Master branch (https://github.com/sunspec/pysunspec)
Step 2 : download (smdx) zip file fom from Master branch (https://github.com/sunspec/models)
Step 3 : copy the smdx file to the folder '/pysunspec-master/sunspec/models/'
Step 4 : download pyserial-3.4 from https://pypi.python.org/pypi/pyserial and Install 
Step 5 : In command prompt naviage to folder with '/pysunspec-master/'
Step 6 : In command prompt Install pysunspec with command 'py -3 setup.py install'

And then i get the Install errors : 

running install
running build
running build_py
running build_scripts
running install_lib
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_all.py to test_all.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_client.py to test_client.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_data.py to test_data.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_device.py to test_device.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_modbus_client.py to test_modbus_client.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_modbus_mbmap.py to test_modbus_mbmap.cpython-36.pyc
byte-compiling C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\models\smdx\manifest.py to manifest.cpython-36.pyc
running install_scripts
running install_egg_info
Removing C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\pysunspec-1.0.8-py3.6.egg-info
Writing C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\pysunspec-1.0.8-py3.6.egg-info
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_all.py", line 50
    print 'Adding local path: ', local_path
                              ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(print 'Adding local path: ', local_path)?

  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_client.py", line 46
    print '*** Failure test_client_device: %s' % str(e)
                                             ^
SyntaxError: invalid syntax

  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_data.py", line 51
    print '*** Failure test_data: %s' % str(e)
                                    ^
SyntaxError: invalid syntax

Sorry: TabError: inconsistent use of tabs and spaces in indentation (test_device.py, line 311)
Sorry: TabError: inconsistent use of tabs and spaces in indentation (test_modbus_client.py, line 58)
  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\core\test\test_modbus_mbmap.py", line 129
    return (count_run, count_passed, count_failed)
         ^
SyntaxError: invalid syntax

  File "C:\Users\user\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\sunspec\models\smdx\manifest.py", line 140
    print xml
            ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(print xml)?

learnetk, 2018-03-28:
Thank you for the help . Atlast got it working. For begineers like me :

Please use the pysunspec from the feature branch https://github.com/sunspec/pysunspec/tree/feature-py2-3

Add the Model description from https://github.com/sunspec/models and copy the smdx folder to pysunspec in '/pysunspec-feature-py2-3/sunspec/models'

Thank you for all the guidence.

altendky, 2018-12-07:
I guess this can be closed?  Though the `manifest.py` over in https://github.com/sunspec/models still needs fixed (https://github.com/sunspec/models/pull/35).  ... and I would have thought the master branch would have been fine for py3 now?

altendky, 2019-07-29:
Also https://github.com/sunspec/pysunspec/issues/60 is still open.  If py3 support is wanted in the mean time https://github.com/altendky/pysunspec works.  Note the extra info at the beginning of the README.