sunspec/pysunspec #65: Drop Python 3.3 support?

altendky, 2019-01-02:
Latest pip [doesn't support Python 3.3](https://github.com/pypa/pip/commit/90b3db476fa3723dd63d51b1d9ea3178addecc69#diff-2eeaed663bd0d25b7e608891384b7298).  CPython support for 3.3 [was dropped in September 2017](https://www.python.org/dev/peps/pep-0398/#lifespan).  Do we need to retain 3.3 support?

This came up over in #46 where I set Travis to update pip (and setuptools and wheel).  That isn't really related to the point of the PR so I've removed it.

toddsnider-epc, 2019-01-02:
I don't know of any compelling reason to maintain support for 3.3.

bobfox, 2019-02-16:
Hi Guys,
I know you are pretty frustrated with me by now and I have to apologize to everyone for my lack of responsiveness on this repository. I just have not had time to review the changes or do anything with it. There are several things going on with SunSpec information models in general that I would like your feedback on. There is a small workgroup preparing some proposed updates. It has been paused due to priorities but is now resuming to complete the work. Anyone who is interested in participating in information model related discussions or reviewing the work, please let me know at bob@sunspec.org. I am going to try to work through the backlog here to get this back on track. The proposed changes will require additional work here if adopted. I really appreciate your contributions and again am sorry for the massive delays on this.
Bob Fox

altendky, 2019-02-16:
@bobfox welcome back!  :]

closed 2020-01-21 by bobfox