sunspec/pysunspec #70: pySunSpec Release 1.0.8 does not suport python3

CharlesGodwin, 2019-04-08:
I followed the instructions in the readme. It sends me to the latest release to download, unzip and install. I do that but use python3 instead of python for my commands. Then I try to run 
`python3 -m unittest discover -v sunspec` Attached are my errors. It looks like python2 compatible code.
Help..

[sunspec.log.gz](https://github.com/sunspec/pysunspec/files/3055609/sunspec.log.gz)

-

mindfuucker, 2019-04-24:
The Sunspec alliance seems to have abandoned this repository. 
This is what I did:
 
1. Downloaded [this fork](https://github.com/altendky/pysunspec) as zip
2. Downloaded models from [this repository](https://github.com/sunspec/models)
3. Inserted the smdx folder (folder, not contents) into the /sunspec/models folder.
4. Replaced the /sunspec/models/smdx/manifest.py file with [this alteration](https://github.com/sunspec/models/pull/35/files)
5. Open terminal inside the root folder
6. sudo python3 setup.py install

Please correct me if im wrong.

CharlesGodwin, 2019-04-24:
Thanks. I've done about the same thing using the original SunSpec repository for testing. I may adapt this repo when I get serious.

However even the models  smdx folder from sunspec are dated and seem incomplete.

bobfox, 2019-04-24:
SunSpec has not abandoned the repository but we have not had the resources to respond to the issues. Also, work has been underway to propose extending the SunSpec information modeling which will cause a major revision to the library. We indicated a few months ago we would be adding resources to address this and are in the process of doing that now. SunSpec is committed to maintaining and enhancing the library and look forward to working with you on it.

mindfuucker, 2019-04-24:
Thank you bobfox for responding. 
My main point of critique towards SunSpec is not accepting pull requests. People have written excellent code to fix the problems. This only takes a single click to merge. 
You could also give another contributor rights to do this. Open Source is about trust, trusting people to do whats right for the project.

bobfox, 2019-04-24:
The problem for SunSpec on this specific topic is that there has been work to resolve these issues as well but possibly done in a different way. The desire was to reconcile the approaches rather than just accept all the pull requests. Unfortunately, this approach does take more time but does not serve the current github users well. I agree in hindsight something different should probably have been done.  It was anticipated that the updated modeling would have been introduced by now but priorities have gotten in the way. This is now becoming one of our highest priorities so you will see progress shortly in a number of different ways. I hope you will hang in there and participate in the effort moving forward.

thiagowiezbicki, 2019-04-30:
> The Sunspec alliance seems to have abandoned this repository.
> This is what I did:
>
> 1. Downloaded [this fork](https://github.com/altendky/pysunspec) as zip
> 2. Downloaded models from [this [...]

Please, if possible commit the updated model list in the official repository. Unfortunately looks trhat Sunspec abandonated the official repository, but we can update and correct the pysunspec bugs. Is it possible?

altendky, 2019-06-25:
> The problem for SunSpec on this specific topic is that there has been work to resolve these issues as well but possibly done in a different way.

@bobfox
Conduct that work and discussion publicly and get it out promptly.  We can help...  but not if it is kept secret.  I have tried very hard to keep my present PRs single-purpose and not in conflict with the rest of the lib or each other.

For all:
Please do not use `sudo` or otherwise run as root when installing Python projects.  Do not run `setup.py` directly.  Mostly don't run your system provided `pip` at all (that means also not with `--user`).  Work in virtualenvs or venvs and use the `pip` they provide to install.  https://bit.ly/py-env and see my updated instructions in #45.

Keep in mind that despite claiming py3 support and having mostly implemented it, afaict you can't actually connect to a device with py3.  Hence #60.

I commented at https://github.com/sunspec/models/issues/39#issuecomment-490526051 about my forks of the `pysunspec` and `models` repos.  They also provide a brief description in their readmes.

https://github.com/altendky/pysunspec
https://github.com/altendky/models