sunspec/pysunspec #73: I don't see a way to remove pysunspec from my Linux Debian system

RDKWyo, 2019-06-03:
How do I completely remove the pysunspec library from my Debian Linux system.  I don't see an uninstall script....RDK

alvaronhs, 2019-06-07:
Read this: https://stackoverflow.com/questions/1550226/python-setup-py-uninstall

RDKWyo, 2019-06-08:
Thanks for the link.  However, as I have searched on this topic, I have learned that it is customary for the developer to supply not only an install process BUT also an uninstall process.  Less chance for beginners like me to screw up the OS....RDK

altendky, 2019-06-25:
The solution is to work in virtual environments using the `pip` they provide and not install into your OS at all (neither with sudo/root nor `--user`, not with `easy_install` nor with `setup.py` directly).  See bit.ly/py-env for a quick overview of virtual environments.

Sadly lots of libraries provide bad documentation on how to install things, including this one.  See #45.