sunspec/pysunspec #74: Is handling of NaN values in scope of spec/library?

andig, 2019-06-28:
I‘ve noticed that SMA inverters use maximum unsigned values or minimum signed values in the respective data type number to indicate a NaN value. For example, model 101 single phase inverters will do so for phases B and C. This behaviour is documented by SMA.

Is this behaviour covered by the sunspec standard and this library? Couldn‘t find it in either code or specs I‘ve looked at?

altendky, 2019-06-28:
On page 19 of the SunSpec information models document it covers unimplemented registers.  On read the SunSpec specified unimplemented value should be returned.  On write the value is thrown out and there is no exception.  Each data type defines its unimplemented value.  For example, not implemented for a uint16 is 0xFFFF (65,535) while for int16 it is 0x8000 (-32768).

andig, 2019-06-28:
Thank you for confirming- this is in line with what the SMA inverters apparently are doing. Is this covered by pysunspec?

altendky, 2019-06-28:
The values are defined here:

https://github.com/sunspec/pysunspec/blob/5242ea9e4c687061cc0b8de4417b1943ed8f9264/sunspec/core/suns.py#L62-L80

and there are a few layers but those end up being used in two places (close to each other) via the `.is_impl()` calls:

https://github.com/sunspec/pysunspec/blob/5242ea9e4c687061cc0b8de4417b1943ed8f9264/sunspec/core/client.py#L361-L362

https://github.com/sunspec/pysunspec/blob/5242ea9e4c687061cc0b8de4417b1943ed8f9264/sunspec/core/client.py#L376-L378

andig, 2019-06-28:
Great, thank you! I had done a quick check for NaN and didn't notice. I'll try to cover the same in gosuncpec.

altendky, 2019-06-28:
Hmm, I haven't dug into the extra layers but `0x7fc00000` is an interesting way to type NaN for the float case.  I guess I would `float('nan')` for py2/3 compatibility and then if the integer is really needed, which seems a bit off, then encode with the `struct` module and...  decode again to get the integer.  Anyways, that's probably not a high priority to refactor.  :]