sunspec/pysunspec #75: Fronius Modbus Card returns empty objects

arturioxas, 2019-07-10:
Been trying this library with an embedded device containing openWRT operating system. I could succesfully execute:
```
>>> import sunspec.core.client as client
>>> d = client.SunSpecClientDevice(client.RTU, 1, '/dev/ttyPORT1')
```
But when asking to print it all, I only get empty objects:
```
>>> print d

model_1 (1):

model_113 (113):

model_120 (120):

model_121 (121):

model_122 (122):

model_123 (123):

model_160 (160):
```

I have double checked that the device is in fact configured in Modbus RTU mode, 9600, 8N1. I have also checked that Fronius doesn't provide a different device id for sunspec-compatible readings.

Could you give a piece of advice on why this could happen?

altendky, 2019-07-10:
Try `d.model_1.read()` then `print d`.  Also, consider using Python3 if possible once #60 is resolved.

arturioxas, 2019-07-12:
Tried a different approach. Made a ser2net connection to a remote desktop. Got the results back succesfully. It doesn't make it clear why something went wrong, but I got it to work. So, one could probably close the issue

altendky, 2019-07-12:
@arturioxas did you try reading the models? I got the same result as you before reading and then after reading a model it had values shown when printing the device.