sunspec/pysunspec #83: cloning with --recursive option resulting older than the newest version of models submodule

marpi82, 2019-11-30:
I have cloned pysunspec with command:
"$ git clone --recursive https://github.com/sunspec/pysunspec.git"

But i don't know why i have older version of submodule (models), newest version have:
```
...except Exception as e:
```
older one which was cloned with --recursive  have:
```
...except Exception, e:
```

altendky, 2019-11-30:
Git submodules are always pinned to a specific version, in this case it is sunspec/models@94a0d35a2db8f011f88d1fa62424046bc11709b8 from...  uh...  November 21, 2016.  There should probably be some testing and then a PR to update this.

marpi82, 2020-01-24:
It is possible to update models link to a newer one? sunspec/models@94a0d35 is quite old as altendky said

altendky, 2020-01-24:
FYI, you can `git -C sunspec/models checkout master` after you clone.  If there are model updates later you can pull them from the submodule directory as well.

closed 2026-08-17 by marpi82