sunspec/pysunspec #93: error in scan with pyhon 3.7.2

RiccardoBa, 2020-02-07:
When I try to connect an RTU device during the scan I get an error because the instruction does not work. data[:4] is a string and then also if the base_addr is right systems not recognize it.
`if data[:4] == b'SunS':
    self.base_addr = addr
    # print('device base address = %d' % self.base_addr)
    break`

I've tried to change b'Suns' in 'Suns' but I get an error in a following statement.
This is my python code
`import sunspec.core.client as client
d = client.SunSpecClientDevice(client.RTU, 1, 'com9', baudrate=19200, timeout=10)`

address, Com, baudrate and timeout and connection using sunspec dashboard works correctly.

altendky, 2020-02-07:
This strikes me as a duplicate of #60.  Maybe give #62 a try (`yourenv/bin/pip install git+https://github.com/altendky/pysunspec@60-altendky-bytes_for_both_py2_and_py3`).

bobfox, 2020-02-14:
Updated with a fix. Give it a try and see if it works for you.