sunspec/pysunspec #99: NoneType attribute label exception on Skybox inverter query

rpdrewes, 2020-11-11:
Running suns.py on my Outback Skybox inverter starts to retrieve good data, but then gets an exception:

```
      [many lines...]
      External Battery Voltage (V):                        54 V         
      Total DC Current (A):                                 0 A         
      Total Power (W):                                      0 W         
      Set Operation (SetOp):                                1           
      Set Inverter State (SetInvState):                     1           
Traceback (most recent call last):
  File "pysunspec/scripts/suns.py", line 91, in <module>
    if model.model_type.label:
AttributeError: 'NoneType' object has no attribute 'label'
```

If I just put a check in suns.py to continue if the object is None, I can avoid the error and the query keeps getting good data.

This may be the Skybox's fault returning data with some fields not populated, but it might be a good idea to program more defensively and put in a check in the code?

shelcrow, 2020-11-11:
Can you please try with pysunspec2 and see if you get the same error? We are deprecating pysunspec.

rpdrewes, 2020-11-12:
I installed sunspec2 that but it isn't clear how to do an equivalent test yet. Standby.